openxc/vi-firmware #319: Ford Platform USB bootloader...

ozman1, 2015-02-11:
So I had PCBs made, ordered parts, assembled, and have assembled units.  Since I did not purchase, they did not come with the USB bootloader programmed.  Is it available somewhere ?  I assume I need to use FlashMagic to program it.  Also, all the Ford binaries online appear to only be for the USB bootloader and not the native ISP.

tangchihwei, 2015-02-11:
Hi @ozman1, great to hear that you have made it through PCBs, parts, and assembled units! You may want to check out our repo(https://github.com/openxc/openlpc-USB_Bootloader) for loading the USB bootloader. You would need a Olimex JTAG programmer and just follow the instruction in the repo.

ozman1, 2015-02-11:
Hey @tangchihwei ... Yeah don't have a Olimex JTAG but will try to get one.  Nobody has a pure hex image online I can just dump to the LPC ?  The serial ISP is like the easiest for me... Just want to verify hardware at this point....

tangchihwei, 2015-02-11:
@ozman1 Please try this hex file here. https://github.com/openxc/reference-vi/tree/gh-pages/firmware/programming
Let us know if there is any question!

ozman1, 2015-02-11:
Got it !  Worked !  Awesome dude !  Thanks !

peplin, 2015-02-11:
That's awesome that you made the boards - any feedback on the design @ozman1 ?

@tangchihwei thanks for the .hex - if that's the v1.1 build, would you mind attaching to the v1.1 release page for the bootloader? https://github.com/openxc/openlpc-USB_Bootloader/releases/tag/v1.1

ozman1, 2015-02-11:
...from a layout / parts perspective, basically went together well...... the OBD connector in the parts list is not available that I could find.  I had to use another part I had and solder some bus wire to make it a right angle connector.  Not a biggee.  Also the footprint for F1 seems too small, had to angle the part to allow connection.  Can't comment on the enclosure, didn't make one....

ozman1, 2015-02-12:
@peplin, @tangchihwei...so i dropped on "vi-obd2-firmware-FORDBOARD-ctv7.0.0.bin" to one board and "vi-translated_obd2-firmware-FORDBOARD-ctv7.0.0.bin" to another.  After restart, the first board has LED1 solid RED.  The second board is the same, but LED1 does blink off occasionally. Not activity on the Android app when either board is plugged into a vehicle.  Any quick thoughts before I start digging into code ?  Thanks for your help :)

tangchihwei, 2015-02-12:
Hi @ozman1 could you try to use vi-translated-obd2-firmware-FORDBOARD from v6.0.3 here https://github.com/openxc/vi-firmware/releases

Additionally, it would be nice if you could reopen this issue, so that it

ozman1, 2015-02-12:
I get the same result....

zacnelson, 2015-02-12:
@ozman1 what vehicle are you plugging into?

ozman1, 2015-02-12:
None at the moment...I just reprogrammed it.

zacnelson, 2015-02-12:
Okay, the 2 firmware files you tested are programmed to go to "sleep" if there is no CAN bus activity or no responses from OBD-II requests on CAN. See here: https://github.com/openxc/vi-firmware/blob/master/src/vi_firmware.cpp#L87. There needs to be active CAN traffic on the vehicle CAN bus in order for these 2 firmware types to function properly.

If you want to test Bluetooth connectivity, you can flash the Emulator firmware and then you should be able to connect over Bluetooth without needing to be connected to an active CAN bus.

ozman1, 2015-02-12:
Gotcha.  Well when I initially reported my issue, I had connected the board in a Ford and a Toyota.  Still only saw the red led come on.  I will go check out this build... back in a bit...

ozman1, 2015-02-12:
@zacnelson ... so only have access to Nissan Titan at the moment... but behavior is definitely improved.  When inserted initially LED toggles slowly between RED and OFF.  When key is inserted in IGN, get a BLUE.  I am assuming CAN is active.  Was I supposed to get a GREEN ?  (based on looking at the referenced code...BLUE is associated with the UART ? ).

zacnelson, 2015-02-12:
@ozman1 that's great news! It sounds like the VI is getting a response from the vehicle. Blue is the correct LED behavior. 

The 2nd LED is for the connection type. GREEN means USB and BLUE means UART (Bluetooth).

ozman1, 2015-02-12:
@zacnelson gotcha...great !  thats a step closer !  So is there a build issue with 7.0.0 ?  ok, so i did have the OpenXC Enabler app up on my Android phone, and I did not see the board...thoughts ?

zacnelson, 2015-02-12:
@ozman1 not sure about the build issue. In order to see the VI in the Enabler, you'll need to pair over bluetooth first. Try the Android smoke test here: http://openxcplatform.com/android/getting-started.html

zacnelson, 2016-04-05:
hey @ozman1 it's been about a year since we last discussed this issue. Did you ever get your boards working properly? Should we close out this issue?

closed 2019-02-11 by emarsman