openxc/vi-firmware #321: Updated Ubuntu; now can't find fabfiles?

Inverted22, 2015-06-15:
Ugh.  Newbie here.  I had the "Install Vagrant (Recommended for All Platforms)" instructions from the OpenXCPlatform site working, however today when I opened my Git Shell I was prompted that there was newer version of Ubuntu available.  Like an idiot, I went ahead and installed it.  Now when I try the "fab reference build" command from the vi-firmware directory within vagrant ssh, I get a "Fatal error: Couldn't find any fabfiles!"  Any ideas what I am doing wrong?  I don't know if I screwed something up in the Ubuntu upgrade or if that is unrelated or what.  Thanks in advance for your help!

Inverted22, 2015-06-15:
More info: when loading vagrant, I notice that the "Mounting shared folders..." section fails with the following:

-----
Failed to mount folders in Linux guest. This is usually because
the "vboxsf" file system is not available. Please verify that
the guest additions are properly installed in the guest and
can work properly. The command attempted was:

mount -t vboxsf -o uid=`id -u vagrant`,gid=`getent group vagrant | cut -d: -f3` vagrant /vagrant
mount -t vboxsf -o uid=`id -u vagrant`,gid=`id -g vagrant` vagrant /vagrant

The error output from the last command was:

stdin: is not a tty
mount: unknown filesystem type 'vboxsf'
-----

Any ideas?

peplin, 2015-06-15:
I think the quickest way to get it working again is to destroy and rebuild the VM with the original Ubuntu version: 

```
vagrant halt
vagrant up
```

I haven't tested the build scripts with any newer versions of Ubuntu, so there's obviously some issues that would need to be worked out if we were to update.

Inverted22, 2015-06-15:
Thanks Christopher.  I tried your suggestions but still get the same "Failed to mount folders in Linux guest" after running "vagrant up."  Any other ideas?  Sorry for the inconvenience.

Inverted22, 2015-06-15:
I tried repairing Vagrant by re-running the installation file (vagrant_1.7.2.msi) but no change in "Failed to mount folders..." error.

peplin, 2015-06-15:
Did you happen to update VirtualBox on your host?

Inverted22, 2015-06-15:
Got it to work again.  I more carefully read your earlier feedback and used the command "vagrant destroy" followed by "vagrant up."  Things seem to be working now (or at least I can complete the tutorial steps once again).  Thank you!!!

peplin, 2015-06-15:
Ah, that's right! I mis-typed when I said just the `halt`. The `destroy` is what actually erases it. Glad it's working!

closed 2015-06-15 by peplin