openxc/vi-firmware #322: upload_hex.bat when path contains spaces

NRyan115, 2015-07-08:
Not sure the proper way to report this issue or solution but hopefully this reaches the right person/people.

Using the batch file included in vi-firmware-7.0.0 (upload_hex.bat)
I had trouble executing the script because, like most Windows users, my folder names contain spaces. The script would not execute / recognize my path correctly EVEN THOUGH I did use quotes around the path. I tried several variations in the command line but discovered that the script itself is not handling input parameters correctly.

The script is using %1, %2 to get parameters but this method seems to terminate a parameter at the first space encountered (regardless of quotes). Instead, %~1, and %~2 should be used which grabs the entire quote as a parameter and then strips the quotation marks.

Below is the updated code that has worked for me on Windows 8.1
I imagine this could be a problem (and solution) for other Windows users.

```
@ECHO OFF

set AVRDUDE=
for %%e in (%PATHEXT%) do (
  for %%X in (avrdude%%e) do (
    if not defined AVRDUDE (
      set AVRDUDE=%%~$PATH:X
    )
  )
)

if not defined AVRDUDE (
  IF EXIST .\\avrdude.exe ( set AVRDUDE=.\\avrdude.exe )
)

if not defined AVRDUDE (
	echo Cannot find avrdude.exe.  Make sure it is in your path or the same directory as avrwrite.bat
	exit /B
)

set AVRDUDE_CONF=../conf/avrdude.conf

if "%~1"=="" (
	echo Firmware file must be passed as first argument, e.g.:
	echo upload_hex.bat firmware.hex com1
	exit /B
)

if "%2~"=="" (
	echo COM port must be passed as second argument, e.g.:
	echo upload_hex.bat firmware.hex com1
	exit /B
)

IF EXIST %~1 (
	set HEX_FILE=%1
	set MCU=32MX795F512L
	set AVRDUDE_ARD_PROGRAMMER=stk500v2
	set AVRDUDE_ARD_BAUDRATE=115200
	set PORT=%~2
	set AVRDUDE_COM_OPTS=-q -V -p %MCU%
	set AVRDUDE_ARD_OPTS=-c %AVRDUDE_ARD_PROGRAMMER% -b %AVRDUDE_ARD_BAUDRATE% -P %PORT%
	echo %AVRDUDE_COM_OPTS%
	echo %AVRDUDE_ARD_OPTS%
	%AVRDUDE% -C %AVRDUDE_CONF% %AVRDUDE_COM_OPTS% %AVRDUDE_ARD_OPTS% -U flash:w:%HEX_FILE%:i
)

IF NOT EXIST "%~1" (
	echo "%~1" doesn't exist
	exit /B
)
```

peplin, 2015-07-08:
Thanks! You found the right place to report the issue. Since you have updates for the code, would you mind committing the changes in a fork of the repository on GitHub and sending a pull request? That would make it easier to test and merge in your changes (and give you credit in the history of changes).

NRyan115, 2015-07-08:
Hello, thanks for getting back to me. I have a version of updated code that worked for me but was not properly tested (for example, I did not try to type in a bad or non existent firmware file, etc). It should be properly tested before committed. Also, I cannot do as you request because I have no real idea what forking or making a pull request means or entails. I simply downloaded the code through my browser and solved this issue when I came to it. I am aware that github is a more powerful tool for share coding but I've never done it and it is not how I am currently using it.

emarsman, 2015-07-08:
Totally understand where you're coming from! It is not that difficult to pickup though. Here is a good starting point: http://openxcplatform.com/contributors/index.html. One link near the bottom points to http://openxcplatform.com/contributors/guide.html which gives step by step instructions.

As to testing, part of the pull request will be testing your code before it goes live. 

Thanks for your contribution!

peplin, 2015-08-30:
For anyone Googling, check out the sample script here for how to upload a hex file that includes spaces. In the meantime, I'm going to close this issue - if anyone can test this and send a PR, we would be happy to include it in the repository.

closed 2015-08-30 by peplin