openxc/vi-firmware #323: No fab files

Craxer01, 2015-07-14:
Hello everyone.
I am just starting withal the configurations and downloads.
Right now, I run the fab reference build

vagrant@precise32:~$ fab reference build
 the result is: Fatal error: Couldn't find any fabfiles!

Now,
vagrant@precise32:~$ ls 
  returns nothing. So how do I make vagrant@precise32:~$  point to the folder where I have the fabfile.py ?

Thank you in advance.

zacnelson, 2015-07-14:
Hi Craxer01, 

You need to navigate into the `/vagrant` directory as outlined in these directions: http://vi-firmware.openxcplatform.com/en/master/getting-started/default-build.html

That should fix your issue. If not we can reopen the issue.

closed 2015-07-14 by zacnelson

Craxer01, 2015-07-14:
hello Zac. Thanks for the answer.
What I am saying is that running the shell I get access to theC:....... /vagrant directory but fab will not work.
Running the vagrant ssh will not give me access to the /vagrant directory as it only shows one .sh file.

emarsman, 2015-07-16:
Hi Craxer,

If you don't have a /vagrant directory or a 'vi-firmware' link in /home/vagrant/, then your 'vagrant up' didn't work right. Go back and look at the logs or look at doing 'vagrant provision' again or 'vagrant destroy' and start over. 

What is the .sh file you're seeing? 

If you're still having issues, please check the threads on the OpenXC google group and post a question there if you need more help.

https://groups.google.com/forum/#!forum/openxc

Regards,
Eric