openxc/vi-firmware #324: emulator firmware is not detecting in android mobile

vijiyasam09, 2015-08-25:
only bluetooth connection is enabled.It shows openxc hardware availability .There is no data updates in openxc enbler application.Also there is no blue light blinking in hardware.Continues red light is availabe.

emarsman, 2015-08-25:
This sounds related to issue [236](https://github.com/openxc/openxc-android/issues/236) in openxc-android. We are investigating. I would recommend rolling back to openxc-android [v6.0.3](https://github.com/openxc/openxc-android/releases/tag/v6.0.3). Please close this issue if that resolves the problem.

peplin, 2015-08-30:
I'm going to close this in favor of openxc/openxc-android#236, since that is almost certainly the root of the issue - an updated Enabler app should be out shortly that works, and in the meantime you can use v6.0.6.

closed 2015-08-30 by peplin