openxc/vi-firmware #327: OpenXC VI is not entering in Programming Mode

DebAndroidDeveloper, 2015-10-16:
Hi,
I already posted this issue in google group,but did not get any response.
I installed openxc python library and required tools and when i run command openxc-control version, i am getting this error in terminal 
"openxc.sources.base.DataSourceError: No USB vehicle interface detected - is one plugged in?"

What could be the reason?Please help me out, i have been struck for 3 days.

The openxc VI device was connected to my Macbook using a usb cable and i flashed the device(copied "vi-obd2-firmware-FORDBOARD-ctv7.0.1"),the green LED is on.I tested with the enabler android app,the app is not able to make socketconnection to the VI,throws exception.I opened the case of the VI and pressed the programming button,but the green LED remains ON,i am afraid the bootloader runs,because i can not see LPC1759 drive(the device had bootloader 1.1 version and i was able to flash it first time).The blue LED becomes ON,if VI is connected using bluetooth.Any help is much appreciated. 

Thanks,

Debasis

zacnelson, 2015-10-16:
https://groups.google.com/forum/#!topic/openxc/MtS_XFmEcXQ

zacnelson, 2015-10-16:
@DebAndroidDeveloper I posted a reply on the Google group. That's the best place to handle these types of setup issues.

closed 2015-10-16 by zacnelson