openxc/vi-firmware #331: about vagrant build environment

oneway3124, 2015-12-31:
After I have installed the vagrant on my ubuntu 14.04 in the virtualbox,  I use the command line "fab reference build" to build the vi-firmware code. I find some error as following,
[image]
Thanks in advance if you can help me about setting up the build environment.

peplin, 2015-12-31:
Interesting - I wonder if that package was dropped from the default OS environment. Try running `sudo pip install prettyprint` to get that package, then re-run the `fab` command.

oneway3124, 2015-12-31:
Ok,Thanks! I know that there is some errors about python.
Right now, I have received the openxc board from us.
Although you're not one member of ford but uber, you also tell me about this project.
Thanks a great!
[image]

oneway3124, 2016-01-21:
The compile environment hasnot been done by myself,
[image]

oneway3124, 2016-01-21:
When I use the troubleshooting (http://vi-firmware.openxcplatform.com/en/master/compile/troubleshooting.html),type "git submodule update --init"
[image]

oneway3124, 2016-01-21:
[image]

peplin, 2016-01-21:
Looks like google code is finally gone - I see nanopb moved to Github, so I'll update the `.gitmodules` to point there. If you try making that change yourself and running `git submodule sync`, I think that will switch it to the new URL.

oneway3124, 2016-01-25:
It seems that it is not ok until now. Can you give me some guide?

emarsman, 2016-01-26:
Hi oneway. If you update to the 'next' branch, everything should work per Chris' recent updates. I will release next to master soon after a few more checks.

closed 2016-01-26 by emarsman