openxc/vi-firmware #339: git submodule full vs relative paths after vagrant up

emarsman, 2016-03-16:
Doing the following:

```
$ git clone https://github.com/openxc/vi-firmware
$ cd vi-firmware/
$ vagrant up
```

Gives the following error:
```
$ git status
fatal: Not a git repository: /vagrant/.git/modules/src/libs/AT-commander
```

Therefore, you can't do any git commands in the native windows git shell. 

The .git files in all the /src/libs/<modules> have the gitdir: set to absolute paths. 

```
$ cat .git
gitdir: /vagrant/.git/modules/src/libs/AT-commander

```
Need these to be relative paths. If doing the 'git submodule update --init --recursive' (or git clone --recursive) in windows, gitdir paths are relative.

```
$ cat .git
gitdir: ../../../.git/modules/src/libs/AT-commander
```

A couple possible fixes. Change setup to use git clone --recursive and remove submodule sync/init from bootstrap? Have the vagrantfile run the commands instead? Are there some git config options to fix this? 

Refs:
https://github.com/Automattic/vip-quickstart/issues/297

http://stackoverflow.com/questions/10953953/ensuring-relative-git-paths

emarsman, 2016-03-16:
Also, some users have issues getting submodules to init correctly due to possible firewall/proxy issues in vagrant. This might alleviate some of that as well.

cwbaldwin, 2019-10-30:
Fixed in #415

closed 2019-10-30 by cwbaldwin