openxc/vi-firmware #344: Add ability to set a PIN required for Bluetooth connections

mitchellrj, 2016-04-12:
I'm deeply concerned that currently any device can connect to the OpenXC VI without authentication of any kind.

peplin, 2016-04-12:
Hey Mitchell, there is no custom PIN support implemented at the moment. To be clear, it does use Bluetooth authentication and encryption, but using the simple pairing protocol (or a default PIN as is common with Bluetooth). The data channel is encrypted. However as you point out, if the VI is powered on and your device is not connected, another device could connect and pair. 

The fact that this i a research and development platform not indented for production use, and the attack window is fairly limited in time and space, I don't judge this to be a high risk. I think it's more likely a rouge app on your own phone could take over the Bluetooth connection and try to read or write to the stream.

Security has not been a priority here because again this is for R&D only - this is the primary reason why raw CAN reads and writes are [disabled by default](http://vi-firmware.openxcplatform.com/en/master/advanced/lowlevel.html#writing-to-can).

pjt0620, 2019-12-20:
As of #425 you can now set the pin used by bluetooth via the build flag DEFAULT_BLUETOOTH_PIN.

closed 2019-12-20 by pjt0620