openxc/vi-firmware #375: Large diagnostic response payloads aren't sent properly in JSON

zacnelson, 2016-09-27:
Initially created an issue in openxc-android [here](https://github.com/openxc/openxc-android/issues/261). Turns out I was able to recreate the same issue with the Python command line tools, so now I think it's an issue in the firmware.

In IsoTp and UDS-C we allow a max diagnostic payload size of 255 bytes. When trying to send many bytes through in the diagnostic response, part of the JSON string is overwritten by other information. Like this:

```json
{"bus":1,"id":2016,"mode":17,"success":true,"pid":2,"payload":"0x030405060708090a0b0c0d0e0f101112131415161718191a1b1c1d1e1f202122232425262728292a2b2c2d2e2f303132333435363738393a3b3c3d3e3f404142434445464748494a4b4c4d4e4f505152535455565758595a5b5c5d5e5f60616{"command_response":"diagnostic_request","status":true}
```

In this example case, the payload should have contained 128 bytes (256 ascii characters when converted to JSON string).

peplin, 2016-09-27:
I'll bet it's hitting the [`MAX_OUTGOING_PAYLOAD_SIZE`](https://github.com/openxc/vi-firmware/blob/master/src/pipeline.h#L24):

```
#define MAX_OUTGOING_PAYLOAD_SIZE 256
```

in pipeline.h, when calling `serialize`, it passes in a buffer of this size as the maximum allowed serialization output.

peplin, 2016-09-27:
The payload in your example stop exactly at 256 bytes - I'll bet the rest is garbage from other regions of memory that we unintentionally read because the payload is not null-terminated.

zacnelson, 2016-09-27:
Yeah, I was looking at that earlier. I did increase that size which helps. But also needed to increase the size of the Queue here: https://github.com/openxc/vi-firmware/blob/0ee0b5ab03174a8729ca417d101848584b26b993/src/util/bytebuffer.h#L7

zacnelson, 2016-09-27:
The queue availability is checked here: https://github.com/openxc/vi-firmware/blob/master/src/pipeline.cpp#L46. It's a bit tough to debug though because adding any debug statements cause a recursive function call to `sendMessage`. Figured that out the hard way...

peplin, 2016-09-27:
Alternatively, you could fit a lot more data in the string if it uses base64 encoding instead of hex, but that would be a breaking API change.

zacnelson, 2016-09-28:
Alright, so after further investigation and attempts to get 255 bytes of payload response out over JSON, I don't think we can do it... I tried increasing the queue size to 700 and the `MAX_OUTGOING_PAYLOAD_SIZE` but now I get an `OOM` error when trying to create the JSON object for the payload. Looks like going to the base64 string encoding type might be the only option. OR we could send out a large diagnostic payload response in multiple messages...

zacnelson, 2016-11-07:
Pull request (https://github.com/openxc/vi-firmware/pull/374) has been merged to limit the diagnostic payload response size to 127 bytes so this error shouldn't happen. Changed this from a `bug` fix to a `feature` enhancement for future development