openxc/vi-firmware #384: Help with uploading generic build of firmware

scarabdesign, 2017-04-12:
I recenlty got a CrossChasm C5 BLE device and I'm attempting to flash it as per the instructions on the [quick guide](http://vi-firmware.openxcplatform.com/en/master/_static/QuickStart%20guide%20to%20using%20C5%20Hardware%20and%20OpenXC.pdf). The version of avdude is 6.1 and the doc refers to 5.10. When I run this command:
```
sudo avrdude -U flash:w:vi-translated_obd2-firmware-CROSSCHASM_C5_BLE-ctv7.2.0.hex -c stk500 -p 32MX795F512L -P /dev/ttyACM0
```
I'm getting the following error:
``` 
avrdude: AVR Part "32MX795F512L" not found.
Valid parts are:
  uc3a0512 = AT32UC3A0512
  c128     = AT90CAN128
  c32      = AT90CAN32
...
```
The config file for avrdude is at /etc/avrdude.conf and I've tried copying it to the location of my hex file also with no avail. A grep of that file reveals that indeed the "part" number 32MX795F512L does not exist in that file. How can I get an update avrdude.conf or get an idea of how I can put in the info myself? Is there other components I'm missing? 

Environment: Linux version 4.9.0-0.bpo.1-amd64 (debian-kernel@lists.debian.org) (gcc version 4.9.2 (Debian 4.9.2-10) ) #1 SMP Debian 4.9.2-2~bpo8+1 (2017-01-26)
The machine is an Intel NUC Skull Canyon

scarabdesign, 2017-04-12:
I found another avrdude.conf that references 32MX795F512L here:
https://github.com/UECIDE/pic32/blob/master/cores/chipKIT-Git/tools/linux64/avrdude.conf
Is it safe to use this one? Do I need to build and install that whole repo?

scarabdesign, 2017-04-12:
Well, I gave up trying with avrdude 6.1 and compiled and installed the 5.10 packaged with the Open-XC-ChrossChasm-Flash package and unfortunately had to blow over my installation of 6.1. I have gotten past this issue now, but later I'll have to go back and see if I can update the 6.1 config with the proper parts for Open-XC. Closing this issue now.

closed 2017-04-12 by scarabdesign

emarsman, 2017-04-12:
Hi,

Yes, we recommend using 5.10. We also tried to get a newer version working, but couldn't get it all right. I can't recall if it was even a v6 or a newer v5. Regardless, we include the 5.10 source with instructions for installing on Mac/Linux. See the download link in the pdf you mentioned.

If you get it working with a newer version, we can test it on all of our PIC platforms and update the docs. I don't think we found the link (https://github.com/UECIDE/pic32/blob/master/cores/chipKIT-Git/tools/linux64/avrdude.conf) you posted. Seems like that would be a start.

Regards,
Eric