openxc/vi-firmware #386: vagrant cffi install issues

emarsman, 2017-06-07:
https://groups.google.com/forum/#!topic/openxc/jfx7EYx1PNU

New incarnation:
==> default:     Running setup.py install for cryptography: finished with status 'error'
==> default:     Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-FlpXZi/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-wu2rhH-record/install-record.txt --single-version-externally-managed --compile:
==> default:     c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
==> default:     compilation terminated.
==> default:     Traceback (most recent call last):
==> default:       File "<string>", line 1, in <module>
==> default:       File "/tmp/pip-build-FlpXZi/cryptography/setup.py", line 321, in <module>
==> default:         **keywords_with_side_effects(sys.argv)
==> default:       File "/usr/lib/python2.7/distutils/core.py", line 112, in setup
==> default:         _setup_distribution = dist = klass(attrs)
==> default:       File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 221, in __init__
==> default:         self.fetch_build_eggs(attrs.pop('setup_requires'))
==> default:       File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 245, in fetch_build_eggs
==> default:         parse_requirements(requires), installer=self.fetch_build_egg
==> default:       File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 576, in resolve
==> default:         dist = best[req.key] = env.best_match(req, self, installer)
==> default:       File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 821, in best_match
==> default:         return self.obtain(req, installer) # try and download/install
==> default:       File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 833, in obtain
==> default:         return installer(requirement)
==> default:       File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 294, in fetch_build_egg
==> default:         return cmd.easy_install(req)
==> default:       File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 608, in easy_install
==> default:         return self.install_item(spec, dist.location, tmpdir, deps)
==> default:       File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 638, in install_item
==> default:         dists = self.install_eggs(spec, download, tmpdir)
==> default:       File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 828, in install_eggs
==> default:         return self.build_and_install(setup_script, setup_base)
==> default:       File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1105, in build_and_install
==> default:         self.run_setup(setup_script, setup_base, args)
==> default:       File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1096, in run_setup
==> default:         raise DistutilsError("Setup script exited with %s" % (v.args[0],))
==> default:     distutils.errors.DistutilsError: Setup script exited with error: command 'gcc' failed with exit status 1
==> default:     
==> default:     ----------------------------------------
==> default: Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-FlpXZi/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-wu2rhH-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-FlpXZi/cryptography/

Can fix by installing libffi-dev (sudo apt-get install libffi-dev).

emarsman, 2017-06-07:
easy fix

closed 2017-06-07 by emarsman

fcojpolo, 2017-06-15:
Sorry, It does not work for me, I´m running "bootstrap.sh"  using a Native Host OS (Ubuntu 16.04) with same problem.

Instead of use only "sudo apt-get install libffi-dev" I used "sudo apt-get install build-essential libssl-dev libffi-dev python-dev" and it works for me.

Source : https://cryptography.io/en/latest/installation/#building-cryptography-on-linux

BR

emarsman, 2017-06-16:
Since you are running directly on Native Ubuntu, did you get the message about using Vagrant on linux (from https://github.com/openxc/vi-firmware/blob/master/script/bootstrap/common.sh#L175)? Vagrant is the preferred environment. 

The remaining packages you list are already installed in the Vagrant environment.