openxc/vi-firmware #402: vagrant fails to load pip

davecramer, 2017-11-02:
easy-install doesn't work as http://pypi.python.org now requires https.

emarsman, 2017-11-03:
I was able to confirm this issue with a new environment. This is the error from vagrant up:

```
==> default: Setting up python-setuptools (0.6.24-1ubuntu1) ...
==> default: Searching for pip
==> default: Reading http://pypi.python.org/simple/pip/
==> default: Couldn't find index page for 'pip' (maybe misspelled?)
==> default: Scanning index of all packages (this may take a while)
==> default: Reading http://pypi.python.org/simple/
==> default: No local packages or download links found for pip
==> default: error: Could not find suitable distribution for Requirement.parse('pip')
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.

```

Do you have a work around? Looking and removing easy_install all together. It will take a bit of time to test though. 

Thanks for reporting the issue.

JTrantow, 2017-12-27:
I was unsuccessful with the original Vagrant development setup. The included Python 2.7.3 included in Ubuntu 12.04 doesn't include pip. I added pip 1.0 install to the provisioning only to find that later in the common.sh script the pip command uses the --pre parameter which must not be supported in oio version 1.0. Updating python/pip in such an old Ubuntu seemed risky so I tried the Ubuntu 14.04 ubuntu/trusty64 box. I was getting some errors with the crypto package and ended up stuck again.

My final attempt was changing to the Ubuntu 16..04 xenial box. The only change I made was config.vm.box = "ubuntu/xenial64" in Vagrantfile. The vagrant up command with this box provisions correctly.

"fab reference build" finishes with "Compiled successfully for FORDBOARD running under a bootloader."

That's as far as I have gone, but it might be easier to update the Vagrant setup with the new box then figure out how to keep an ancient version going.

emarsman, 2018-01-05:
My initial testing of this matches yours. Don't see any errors in firmware or python running on 16.04. The issue I do have is our test (e.g. "fab test") currently uses clang-3.4 on 12.04. 3.5 or 3.8 is available on 16.04, but reports errors. Initial ones are in uds-c submodule. 

These are probably worth going through and fixing, but I'm not sure how many issues there are or how long it will take to work through them. 

Note: clang isn't there by default. Need to do "sudo apt-get install clang" before "fab test". Clang should be added to the bootstrap. Actually, clang is in /script/bootstrap/devel.sh. Need to figure out why that isn't being called too. 

The other item to check is Xenail being supported by our Travis-CI. Looks like Travis doesn't support 16.04 yet, but will shortly.

https://github.com/travis-ci/travis-ci/issues/5821
https://blog.travis-ci.com/2017-12-01-new-update-schedule-for-linux-build-images

pjt0620, 2018-12-05:
Pip has been updated as of #406

closed 2018-12-05 by pjt0620