openxc/vi-firmware #442: Compiling the Default Configuration - Syntax error, using vagrant

tpho216, 2020-02-24:
http://vi-firmware.openxcplatform.com/en/master/getting-started/development-environment.html

I followed the documentation like above and after do the command 'vagrant up' and 'vagrant ssh', I tried to run 'fab reference build' but it says I have to install fabric and, after I did 'sudo apt-get install fabric', when I run 'fab reference build', it gave me an error, 

```
vagrant@precise32:/vagrant/vi-firmware-8.0.0$ fab reference build
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/fabric/main.py", line 594, in main
    docstring, callables, default = load_fabfile(fabfile)
  File "/usr/lib/python2.7/dist-packages/fabric/main.py", line 156, in load_fabfile     
    imported = importer(os.path.splitext(fabfile)[0])
  File "/vagrant/vi-firmware-8.0.0/fabfile.py", line 249
    print(f"Make tests passed: {tests_passed}/2")
                                               ^
SyntaxError: invalid syntax
```

I have not altered anything in the default fabfile.py file in the vi-firmware-8.00 release. Then, what kind of syntax was incorrect and how do I fix this issue.

**P/S: run vagrant terminal on Visual Studio Code**
[image]

**My VirtualBox configuration**
[image]

pjt0620, 2020-02-24:
Hi tpho216, I see you are using an old version of the vagrant virtual machine (Ubuntu 12) the current firmware v8.0.0 does not support old virtual machines as stated in the changelog and release notes,  fortunately this is an easy fix all you have to do re create the virtual machine.
To do this you should just have to run the below commands, be sure if you have created any items outside of the vi-firmware folder in the vm that you back them up before running the commands.
> vagrant destroy
> vagrant up

tpho216, 2020-02-24:
> Hi tpho216, I see you are using an old version of the vagrant virtual machine (Ubuntu 12) the current firmware v8.0.0 does not support old virtual machines as stated in the changelog and release [...]

Hi I tried the following commands but still have issue with the syntax.
[image]

pjt0620, 2020-02-24:
What was the output of those commands? You still seem to have not updated to the new version of the vm.

tpho216, 2020-02-24:
Outputs
[image]
[image]

pjt0620, 2020-02-24:
Hmm Is the folder VI_Reference_firmware up to date with the git repo?

tpho216, 2020-02-24:
> 
> 
> Hmm Is the folder VI_Reference_firmware up to date with the git repo?

[image]

It is, I cloned it from the repo

tpho216, 2020-02-24:
> 
> 
> Outputs
> [image]
> [image]
> [image]

I'll try update to Ubuntu version 14

pjt0620, 2020-02-24:
Thats weird, unfortunatley manually updating it wont help since the new vagrant file also sets up all of the dependencies and libraries needed for compilation. What does your Vagrantfile look like?

tpho216, 2020-02-24:
**I have 2 vagrant files, one inside the folder D:\Workspace\VI_Reference_firmware\ -> Vagrantfile, 
look like this,** 
```
# -*- mode: ruby -*-
# vi: set ft=ruby :

# All Vagrant configuration is done below. The "2" in Vagrant.configure
# configures the configuration version (we support older styles for
# backwards compatibility). Please don't change it unless you know what
# you're doing.
Vagrant.configure("2") do |config|
  # The most common configuration options are documented and commented below.
  # For a complete reference, please see the online documentation at
  # https://docs.vagrantup.com.

  # Every Vagrant development environment requires a box. You can search for
  # boxes at https://vagrantcloud.com/search.
  config.vm.box = "hashicorp/precise32"

  # Disable automatic box update checking. If you disable this, then
  # boxes will only be checked for updates when the user runs
  # `vagrant box outdated`. This is not recommended.
  # config.vm.box_check_update = false

  # Create a forwarded port mapping which allows access to a specific port
  # within the machine from a port on the host machine. In the example below,
  # accessing "localhost:8080" will access port 80 on the guest machine.
  # NOTE: This will enable public access to the opened port
  # config.vm.network "forwarded_port", guest: 80, host: 8080

  # Create a forwarded port mapping which allows access to a specific port
  # within the machine from a port on the host machine and only allow access
  # via 127.0.0.1 to disable public access
  # config.vm.network "forwarded_port", guest: 80, host: 8080, host_ip: "127.0.0.1"

  # Create a private network, which allows host-only access to the machine
  # using a specific IP.
  # config.vm.network "private_network", ip: "192.168.33.10"

  # Create a public network, which generally matched to bridged network.
  # Bridged networks make the machine appear as another physical device on
  # your network.
  # config.vm.network "public_network"

  # Share an additional folder to the guest VM. The first argument is
  # the path on the host to the actual folder. The second argument is
  # the path on the guest to mount the folder. And the optional third
  # argument is a set of non-required options.
  # config.vm.synced_folder "../data", "/vagrant_data"

  # Provider-specific configuration so you can fine-tune various
  # backing providers for Vagrant. These expose provider-specific options.
  # Example for VirtualBox:
  #
  # config.vm.provider "virtualbox" do |vb|
  #   # Display the VirtualBox GUI when booting the machine
  #   vb.gui = true
  #
  #   # Customize the amount of memory on the VM:
  #   vb.memory = "1024"
  # end
  #
  # View the documentation for the provider you are using for more
  # information on available options.

  # Enable provisioning with a shell script. Additional provisioners such as
  # Ansible, Chef, Docker, Puppet and Salt are also available. Please see the
  # documentation for more information about their specific syntax and use.
  # config.vm.provision "shell", inline: <<-SHELL
  #   apt-get update
  #   apt-get install -y apache2
  # SHELL
end
```

**one inside the git repo, D:\Workspace\VI_Reference_firmware\vi-firmware\ -> Vagrantfile** 
```
# -*- coding: utf-8 -*-
# -*- mode: ruby -*-
# vi: set ft=ruby :

# Vagrantfile API/syntax version. Don't touch unless you know what you're doing!
VAGRANTFILE_API_VERSION = "2"

def total_cpus
  require 'etc'
  Etc.nprocessors
end

Vagrant.configure(VAGRANTFILE_API_VERSION) do |config|
  # For a complete reference, please see the online documentation at
  # vagrantup.com.

  # Every Vagrant virtual environment requires a box to build off of.
  config.vm.box = "bento/ubuntu-18.04"
  config.vm.box_version = "201912.03.0"
  config.vm.provider :virtualbox do |v|
    if total_cpus > 1
      v.cpus = total_cpus - 1
	else
	  v.cpus = 1
	end
  end
  
  # Check for proxy enviroment variable and set it
  if ENV['HTTP_PROXY'] || ENV['HTTPS_PROXY']
    if Vagrant.has_plugin?("vagrant-proxyconf")
      if ENV['HTTP_PROXY']
        config.proxy.http = ENV['HTTP_PROXY']
        config.apt_proxy.http = ENV['HTTP_PROXY']
      end
      if ENV['HTTPS_PROXY']
        config.proxy.https = ENV['HTTPS_PROXY']
        config.apt_proxy.https = ENV['HTTP_PROXY']
      end
      if ENV['NO_PROXY'] 
        config.proxy.no_proxy = ENV['NO_PROXY']
      end
    else
      abort("ERROR, vagrant-proxyconf not installed run ‘vagrant plugin install vagrant-proxyconf’ to install it") 
    end
  end

  config.vm.box_download_insecure = true
  config.vm.provision "shell", privileged: false, keep_color: true do |s|
    s.inline = "ln -fs /vagrant vi-firmware;"
    s.inline += "VAGRANT=1 vi-firmware/script/bootstrap.sh"

  end
end

```

pjt0620, 2020-02-24:
Ok now we are getting somewhere, the one at D:\Workspace\VI_Reference_firmware\ is out of date, my guess you are running your vagrant commands from there which is why you are getting the old Ubuntu palagon box, try running the vagrant commands from within the git repo D:\Workspace\VI_Reference_firmware\vi-firmware\.

tpho216, 2020-02-24:
> Ok now we are getting somewhere, the one at D:\Workspace\VI_Reference_firmware\ is out of date, my guess you are running your vagrant commands from there which is why you are getting the old Ubuntu [...]

Thanks, I'll try now

tpho216, 2020-02-24:
Successfully compiled the file. Thanks for your immediate help! We're able to close this issue
[image]

closed 2020-02-24 by tpho216